gdsfactory/sax #4: time domain

joamatab, 2022-02-18:
How could we add time domain simulations to SAX?

similar to [photontorch](https://github.com/flaport/photontorch/)

flaport, 2022-02-19:
I could probably implement something similar to photontorch, but personally i'd like a better implementation.

That said. I think it could be a good starting point. I'll give it a go one of the coming days/weeks when I find the time.

joamatab, 2022-02-23:
Here is some ideas

https://github.com/joamatab/awesome_photonics

joamatab, 2022-02-23:
in particular this repo has a time domain for electronics

https://github.com/TeledyneLeCroy/SignalIntegrity

ChrisDaunt1984, 2022-08-29:
Hi, Is there any progress on this? 
SignalIntegrity is being used, in combination with a in-house photonic models to perform end to end electronic-photonic simulations. Unfortunately, SignalIntegrity supports just linear models for now so "memory" effects like what is possible with Caphe and PhotonTorch are not possible. I would like to move our in-house models to something more scalable like photontorch (which is not that active?) or SAX if it supported time domain so am interested in hearing if there is any progress.

flaport, 2022-08-31:
Hi @ChrisDaunt1984 ,

Photontorch is basically in maintenance mode, I kind of like the SAX approach better (sparse by default, swappable backends, ...). It also integrates better with GDSFactory.

To be honest, I currently don't have the time to work on this, but I have some ideas on how to best implement a simple version (just like the photontorch one). I of course don't know your schedule but if you'd like to implement something like this for SAX, I can give you some guidance and I'd happily give you ownership over it 😉

Let me know if you're interested.

ChrisDaunt1984, 2022-09-02:
Hi @flaport

OK, if photontorch is in limbo, then probably best not to invest time into it. I looked at the implementation and I think I can follow what's going on (just). 

Currently, our scipy/numpy solution works and integrates with SignalIntegrity. Pete will actually demo it at ECOC so I'm am not under immediate pressure, but I am kind of unsatisfied with the scalability so looking at other options. 

Id like to say that I have the time and am interested but realistically, I won't have the time either until after November. Saying that, I am interested in working on this (or contributing where I can)

flaport, 2022-09-03:
sounds good! I'll ping you in November 🙂

joamatab, 2022-09-04:
Maybe we can port the time domain model from [simphony](https://github.com/BYUCamachoLab/simphony) into SAX.

@skandanc
@sequoiap
@cjcarver
@austp
@alexsludds

Jan-David-Black, 2022-11-29:
Hey @ChrisDaunt1984, @flaport, @joamatab,
is there currently work in progress on this, or is it still untouched? Do you see fit for me to help somehow?
Regards
JD

ChrisDaunt1984, 2022-11-29:
Hey @Jan-David-Black 
I have not been able to touch it and will probably now not be able to get around to it in the short term (schedules have changed).
I am still interested in the availability though

joamatab, 2022-11-29:
Now that we have schematic driven layout it would be great to interface with signalIntegrity, which is time domain capable
https://github.com/TeledyneLeCroy/SignalIntegrity

flaport, 2022-11-30:
> Hey @ChrisDaunt1984, @flaport, @joamatab, is there currently work in progress on this, or is it still untouched? Do you see fit for me to help somehow? Regards JD

Hi @Jan-David-Black , 

This issue has not been touched. Feel free to pick it up!

joamatab, 2023-01-19:
Time domain models for memory container blocks

JAX may have a better way to do this

SkandanC, 2023-01-23:
Hello @joamatab,

There is an ongoing project at the CamachoLab where we are trying to port Simphony to JAX, and add time domain capabilities. 

@seqouiap will know more about this,

SkandanC, 2023-01-23:
Hello @joamatab,

There is a project in the CamachoLab where they are trying to write Simphony in JAX, and add time domain stuff

ChrisDaunt1984, 2023-01-23:
Hi

Reading up on JAX related things a bit more and seeing what is available. Most of these "memory" artifacts are usually handled by tracking some form of internal state vs time (charge, temperature, energy... etc). In the case of Caphe and PhotonTorch (I believe) or Verilog-A for electronics, this is handled by numerically integrating an ODE for each time step.

JAX seems to already have at least an experimental ode solver in place which looks like it could be used [https://github.com/google/jax/blob/main/jax/experimental/ode.py](https://github.com/google/jax/blob/main/jax/experimental/ode.py) 

Alternatively, I like the look of [https://github.com/patrick-kidger/diffrax](https://github.com/patrick-kidger/diffrax) where one can step through the integration allowing the input and output optical/electronic signals at the circuit nodes to be updated at each time step 
[https://docs.kidger.site/diffrax/usage/manual-stepping/](https://docs.kidger.site/diffrax/usage/manual-stepping/)

flaport, 2023-01-23:
Indeed, tracking an internal state and integrating over it is the default way of tackling such time dependent problems.

Diffrax seems to be almost exactly what we need for this. Good find! I'll have to play around a bit with it to get to know it a bit better :)