gdsfactory/sax #6: jax as optional dependency

joamatab, 2022-04-24:
Hi Floris,

what do you think of having `jax` as an optional dependency?

that way windows users can still run circuit simulations, 

maybe we could have a mode where sax tries to import `jax` and if it fails you can still do some functions (such as circuit sims)

let me know what you think

flaport, 2022-04-24:
Hey Joaquin,

yes - that has been on my todo list for a while.

In the meantime windows users can use one of the unofficial windows builds for JAX: https://whls.blob.core.windows.net/unstable/index.html

The easiest way to install from this repository is to set the PIP_FIND_LINKS environment variable on windows to that url. After that pip will automatically find the unofficial builds as well.

flaport, 2022-06-14:
Hey @joamatab ,

I gave this a shot, but it seems to be more work that initially anticipated. Quite a bit of the SAX utilities use JAX-only tools and functions. I'm not saying it's impossible but to do it well it would require some serious refactoring...

I might pick this up at a later date though, I'll leave the issue open for that.

In the mean time I think the solution I proposed above should be good enough.

flaport, 2022-06-17:
Hey @joamatab ,

I gave it another shot. It's not the most clean solution but it works! You can now install and use SAX without the JAX-dependency!

check out the [Install Instructions](https://github.com/flaport/sax/#installation) for more info.

closed 2022-06-17 by flaport