gdsfactory/sax #9: simulate hiearachical components

joamatab, 2022-08-09:
How can we simulate hierarchical components?

See for example 2 mzis connected to each other

joamatab, 2022-08-15:
I added an example here

https://github.com/gdsfactory/gdsfactory/blob/master/fixme/circuit_simulations_fixme.py

flaport, 2022-08-16:
This was implemented today: sax>=0.8.0.

Note that this is a breaking change. Your notebooks depending on SAX will likely break. But what we get in return is a much better pydantic-based netlist parser and a better circuit resolver. 

I'll fix the example notebooks in gdsfactory soon.

closed 2022-08-16 by flaport

flaport, 2022-08-16:
see:

- https://flaport.github.io/sax/nbs/06_netlist.html
- https://flaport.github.io/sax/nbs/08_circuit.html

joamatab, 2022-08-16:
Awesome. Thank you Floris