gdsfactory/sax #10: Models of active components

alexsludds, 2022-09-01:
Would it be possible to include models of active components? I want to be able to reference components within the circuit, change an applied voltage and see the effect the phase shift has on the system.

Best,
-Alex

alexsludds, 2022-09-04:
Following up on this, I recently put some time into a really crude example of voltage controlled phase shifter in SAX:
[https://gdsfactory.github.io/gdsfactory/notebooks/plugins/sax/sax.html#Tuning](Voltage Tuning in SAX)

There is still some work to do there to realize easy simulation of circuits with large component counts (1000's of phase shifters), but my next request is that each component has to ability to maintain state.
For example, suppose I want to model an RC type time constant on my thermo-optic phase shifters. I could implement an internal variable which updates on each timestep based on the applied voltage/applied thermal power:
[https://en.wikipedia.org/wiki/Low-pass_filter#Discrete-time_realization](Discrete Time Low Pass Filters)

Given that SAX uses functions as it's way of manipulating objects I was wondering if there is any way of a component maintaining state?

Thanks for everything you guys do, I really love this software,
-Alex Sludds

tvt173, 2022-09-06:
@flaport may have better ideas, but I'll offer a trick I've used in the past for things like this...

you can create a _callable object_ in python, by defining a class with the `__call__` special method. such an object will be initialized and maintain state, but can otherwise be used just like a function
https://python-course.eu/oop/callable-instances-classes.php

maybe something like this will fit your needs 🙂

flaport, 2022-09-06:
Hi @tvt173 , this method is indeed the default for PyTorch and is generally speaking a pretty good approach.

The problem is that JAX functions are supposed to be [pure functions](https://jax.readthedocs.io/en/latest/notebooks/Common_Gotchas_in_JAX.html#pure-functions) and hence they cannot depend on any hidden state (at least if you want to be able to jit them).

Any hidden state that the circuit/network depends on must be given as an argument to the function. This is by the way something that SAX supports as you can specify arbitrary arguments to each model function.

However, we could streamline this process a bit better. Maybe the approach that the neural network library [flax](https://github.com/google/flax) is taking is pretty good. At first sight they seem to be implementing more or less what you're proposing (a class with a __call__ method), however [when you look closer](https://github.com/google/flax#what-does-flax-look-like) it's just some syntactic sugar from what I was describing above: hidden variables are given as first argument to the `apply` method.

tvt173, 2022-09-07:
nice, sounds promising. i couldn't recall if evaluation of the s-matrices of individual components was in the scope of what was getting jitted or not. if it were brought outside outside the scope, that would give us a bit of extra flexibility, but glad to see there is an option even for the more constrained case

alexsludds, 2023-02-08:
Hi folks, I know it's been a while, but I've been stuck on this and wanted to reach out for folks advice.

[I made a fork of sax](https://github.com/alexsludds/sax_with_time_domain) where I've been trying to add handling of __call__ methods to sax. 

To see this in action please clone the linked repo and in the examples folder run the "dynamic_sax_component" python notebook. 
The first half of the notebook is the standard approach we take with gdsfactory today. The second half of the notebook shows an example of a class for a phase shifter with a call method being used with the class responding to changes in voltage. 

The problem with the code I added can be seen in the second to last code snippet, where you see that whenever we call the MZI circuit the class for the thermal phase shifter is being re-initialized each time, meaning it can not store data/posses memory.

I spent a while digging through the sax source code, but have not seen anything I can change which will stop the thermal class being re-initialized.

Hoping for your insights,
-Alex

flaport, 2023-02-09:
Hi Alex, I'll try to take a look today or tomorrow.