gdsfactory/sax #15: Using circuits as models for other circuits

mdecea, 2023-02-16:
I'm trying to build a hierarchical circuit that uses a subcircuit as a model.

Something like:
    top_circuit:
         instances: {- model
                          - sub_circuit}

If I naively try to add a circuit as a model I get the following error message: `AttributeError: 'function' object has no attribute 'items'`

Here is a minimum (non-)working example:

```

import sax
import numpy as np

def ring(length=200.0, coupling=0.1, prop_loss=0.1, neff=1.4, ng=1.4, wl0=1.5, wl=1.5):

    ring_sax, _ = sax.circuit(
        netlist={
            "instances": {
                "dc": {"component": "coupler", "settings": {"coupling": coupling}},
                "top": {
                    "component": "straight",
                    "settings": {
                        "length": length,
                        "loss": prop_loss, 
                        "neff": neff,
                        "ng": ng,
                        "wl0": wl0,
                        "wl": wl,
                    },
                },
            },
            "connections": {
                "dc,out1": "top,in0",
                "top,out0": "dc,in1",
            },
            "ports": {
                "in0": "dc,in0",
                "out0": "dc,out0",
            },
        },
        models={
            "coupler": sax.models.coupler,
            "straight": sax.models.straight,
        },
    )

    return ring_sax

def ring_test(ring_length=10.0, ring_prop_loss=1.0, ring_coupling=0.001, neff=1.4, ng=1.4, wl0=1.3, wl=1.3):

    r_model, _ = sax.circuit(
        netlist={
            "instances": {
                "wg": {
                    "component": "straight",
                    "settings": {
                        "length": 100.0,
                        "loss": 0.1,
                        "neff": neff,
                        "ng": ng,
                        "wl0": wl0,
                        "wl": wl,}
                },
                "ring_ps": {
                    "component": "ring",
                    "settings": {
                        "length": ring_length,
                        "coupling": ring_coupling,
                        "prop_loss": ring_prop_loss,
                        "neff": neff,
                        "ng": ng,
                        "wl0": wl0,
                        "wl": wl,}
                },
            },
            "connections": {
                "wg,out0":"ring_ps,in0",
            },
            "ports": {
                "in0":"wg,in0",
                "out0":"ring_ps,out0",
            },
        },
        models={
            "straight": sax.models.straight,
            "ring": ring,
        },

    )

    return r_model

wl = np.linspace(1.265, 1.285, 1000)
ring_circuit = ring_test(wl=wl) 
ring_circuit()
```

If in the `models` dictionary I do `"ring": ring(),`, then there's no error but the results are not correct (it acts as if there were no ring).

Am I missing something?

Thank you!

tvt173, 2023-02-16:
i believe 
```python
models={
            "straight": sax.models.straight,
            "ring": ring,
        },
```
is correct

but you should
```python
    return ring_sax()
```
(note added parentheses)

at the end of your `ring()` function definition

mdecea, 2023-02-16:
I just tried and it still does not work. It does not result in an error but it acts as if there was no `ring` element in the circuit.

flaport, 2023-02-16:
Hi @mdecea , `sax.circuit` returns a function (which returns an S-matrix), not an S-matrix itself! You can leave out the wrapping functions to make it work:

```python
ring, _ = sax.circuit(
    netlist={
        "instances": {
            "dc": "coupler",
            "top": "straight",
        },
        "connections": {
            "dc,out1": "top,in0",
            "top,out0": "dc,in1",
        },
        "ports": {
            "in0": "dc,in0",
            "out0": "dc,out0",
        },
    },
    models={
        "coupler": sax.models.coupler,
        "straight": sax.models.straight,
    },
)
ring_test, _ = sax.circuit(
    netlist={
        "instances": {
            "wg": "straight",
            "ring_ps": "ring",
        },
        "connections": {
            "wg,out0": "ring_ps,in0",
        },
        "ports": {
            "in0": "wg,in0",
            "out0": "ring_ps,out0",
        },
    },
    models={
        "straight": sax.models.straight,
        "ring": ring,
    },
)

wl = np.linspace(1.265, 1.285, 1000)
ring_test()
```
```
{('out0', 'out0'): Array(0.+0.j, dtype=complex64),
 ('out0', 'in0'): Array(0.7785834+0.5514743j, dtype=complex64),
 ('in0', 'out0'): Array(0.7785834+0.5514743j, dtype=complex64),
 ('in0', 'in0'): Array(0.+0.j, dtype=complex64)}
```

If you really want or need to wrap the resulting circuit functions in a new model function, it's best (for jax-related performance reasons) to leave the creation of the circuit outside the new model function. Something like this would work:

```python
_ring, _ = sax.circuit(
    netlist={
        "instances": {
            "dc": "coupler",
            "top": "straight",
        },
        "connections": {
            "dc,out1": "top,in0",
            "top,out0": "dc,in1",
        },
        "ports": {
            "in0": "dc,in0",
            "out0": "dc,out0",
        },
    },
    models={
        "coupler": sax.models.coupler,
        "straight": sax.models.straight,
    },
)

def ring(length=200.0, coupling=0.1, prop_loss=0.1, neff=1.4, ng=1.4, wl0=1.5, wl=1.5):
    return _ring(
        dc={"coupling": coupling},
        top={
            "length": length,
            "loss": prop_loss,
            "neff": neff,
            "ng": ng,
            "wl0": wl0,
            "wl": wl,
        },
    )

_ring_test, _ = sax.circuit(
    netlist={
        "instances": {
            "wg": "straight",
            "ring_ps": "ring",
        },
        "connections": {
            "wg,out0": "ring_ps,in0",
        },
        "ports": {
            "in0": "wg,in0",
            "out0": "ring_ps,out0",
        },
    },
    models={
        "straight": sax.models.straight,
        "ring": ring,
    },
)

def ring_test(
    ring_length=10.0,
    ring_prop_loss=1.0,
    ring_coupling=0.001,
    neff=1.4,
    ng=1.4,
    wl0=1.3,
    wl=1.3,
):
    return _ring_test(
        wg={"length": 100.0, "loss": 0.1, "neff": neff, "ng": ng, "wl0": wl0, "wl": wl},
        ring_ps={
            "length": ring_length,
            "coupling": ring_coupling,
            "prop_loss": ring_prop_loss,
            "neff": neff,
            "ng": ng,
            "wl0": wl0,
            "wl": wl,
        },
    )
```
```
{('out0', 'out0'): Array(0.+0.j, dtype=complex64),
 ('out0', 'in0'): Array(-0.11200467-0.29557914j, dtype=complex64),
 ('in0', 'out0'): Array(-0.11200467-0.29557914j, dtype=complex64),
 ('in0', 'in0'): Array(0.+0.j, dtype=complex64)}
```

The reason you can provide settings to the netlist on the creation of the circuit is usually just to set defaults. If you want to make it parametrizable, the approach in my second example is advised as it keeps the resulting model function [nice and pure](https://jax.readthedocs.io/en/latest/notebooks/Common_Gotchas_in_JAX.html#pure-functions)

Let me know if you have any more questions.

mdecea, 2023-02-16:
Thanks a lot for the detailed answer @flaport ! I think I got it

closed 2023-02-16 by mdecea