gdsfactory/sax #17: install on windows

joamatab, 2023-06-08:
How can we install sax on windows?

i think the SAX [instructions](https://github.com/flaport/sax#installation) for windows does not seem to work anymore

```
set PIP_FIND_LINKS="https://whls.blob.core.windows.net/unstable/index.html"
pip install sax[jax]
```

https://github.com/gdsfactory/gdsfactory/actions/runs/5208126572/jobs/9396371280

flaport, 2023-06-09:
Hey joamatab, I recently installed sax/jax/jaxlib on windows and it worked perfectly for me. What in those logs makes you think it's related sax/jax? I can't find it :)

joamatab, 2023-06-13:
apparently the latest version of jax 0.4.12 doesnt have a corresponding jaxlib version yet

@skandanc

flaport, 2023-06-13:
I am not pinning jax/jaxlib, so you should still be able to install it, no? Unless another library is pinning it > 0.4...

daquintero, 2023-06-14:
So just to update, I have been doing a few tests in between Linux and Windows.

On Windows, I have clean installed the latest mambaforge version from [here ](https://github.com/conda-forge/miniforge#mambaforge). I cloned the sax repository, and ran the following commands from the README.md from the bash terminal:

```
$ python --version
Python 3.10.10

$ set PIP_FIND_LINKS="https://whls.blob.core.windows.net/unstable/index.html"
(base) 
dario@DESKTOP-1T2BQKV MINGW64 ~/Documents/phd/sax (main)
$ pip install -e .[jax]
Obtaining file:///C:/Users/dario/Documents/phd/sax
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... done
  Preparing editable metadata (pyproject.toml) ... done
Requirement already satisfied: tables in c:\users\dario\mambaforge\lib\site-packages (from sax==0.8.7) (3.8.0)
Requirement already satisfied: natsort in c:\users\dario\mambaforge\lib\site-packages (from sax==0.8.7) (8.3.1)
Requirement already satisfied: pyyaml in c:\users\dario\mambaforge\lib\site-packages (from sax==0.8.7) (6.0)
Requirement already satisfied: h5py in c:\users\dario\mambaforge\lib\site-packages (from sax==0.8.7) (3.8.0)
Requirement already satisfied: numpy in c:\users\dario\mambaforge\lib\site-packages (from sax==0.8.7) (1.24.3)
Requirement already satisfied: orjson in c:\users\dario\appdata\roaming\python\python310\site-packages (from sax==0.8.7) (3.8.12)
Requirement already satisfied: pydantic<2 in c:\users\dario\mambaforge\lib\site-packages (from sax==0.8.7) (1.10.9)
Requirement already satisfied: tqdm in c:\users\dario\mambaforge\lib\site-packages (from sax==0.8.7) (4.65.0)
Requirement already satisfied: fastcore in c:\users\dario\mambaforge\lib\site-packages (from sax==0.8.7) (1.5.29)
Requirement already satisfied: networkx in c:\users\dario\mambaforge\lib\site-packages (from sax==0.8.7) (3.1)
Requirement already satisfied: black in c:\users\dario\mambaforge\lib\site-packages (from sax==0.8.7) (23.3.0)
Collecting flax
  Using cached flax-0.6.10-py3-none-any.whl (226 kB)
Collecting jax
  Using cached jax-0.4.12.tar.gz (1.3 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
ERROR: Could not find a version that satisfies the requirement jaxlib; extra == "jax" (from sax[jax]) (from versions: none)
ERROR: No matching distribution found for jaxlib; extra == "jax"
(base) 
```

However, if I do:
```
$ pip install -e .
Obtaining file:///C:/Users/dario/Documents/phd/sax
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... done
  Preparing editable metadata (pyproject.toml) ... done
....
Requirement already satisfied: msgpack in c:\users\dario\mambaforge\lib\site-packages (from blosc2~=2.0.0->tables->sax==0.8.7) (1.0.5)
Building wheels for collected packages: sax
  Building editable for sax (pyproject.toml) ... done
  Created wheel for sax: filename=sax-0.8.7-0.editable-py3-none-any.whl size=7075 sha256=663b80d6086cc1ac9755d7492970857cc614687177220343193a78cdfb7ec943
  Stored in directory: C:\Users\dario\AppData\Local\Temp\pip-ephem-wheel-cache-hltv24y_\wheels\ed\e9\e4\6df96d24ff27f0c79f7086ec1819ea44b546aa0bece6d632de
Successfully built sax
Installing collected packages: sax
  Attempting uninstall: sax
    Found existing installation: sax 0.8.7
    Uninstalling sax-0.8.7:
      Successfully uninstalled sax-0.8.7
Successfully installed sax-0.8.7
(base) 
```

If I run one of your examples to test the windows installation, say 0.7_layout_aware, if I uncomment all the jax imports then:

```
import sax
C:\Users\dario\Documents\phd\sax\sax\backends\default.py:23: UserWarning: [NO JAX] skipping jit! Please install JAX!
  warnings.warn("[NO JAX] skipping jit! Please install JAX!")

...

>>> mzi1()
{('o1', 'o1'): 0.0,
 ('o3', 'o3'): 0j,
 ('o3', 'o2'): 0j,
 ('o2', 'o3'): 0j,
 ('o2', 'o2'): 0j,
 ('o1', 'o3'): (0.5000000000000001+0.5000000000000001j),
 ('o1', 'o2'): (0.5000000000000001+0.5000000000000001j),
 ('o3', 'o1'): (0.5000000000000001+0.5000000000000001j),
 ('o2', 'o1'): (0.5000000000000001+0.5000000000000001j)}
```

Maybe hte source of the problem is in the installation instruction, so that it just has to be `pip install sax` rather than `pip install sax[jax]`. This said, I am running JIT/JAX from a Linux server, but I think this may explain some of the issues for those doing the windows installations.

daquintero, 2023-06-14:
Haven't explored this much but out of curiousity also ran this on Windows, on the `jax` and `jaxlib` separate installation:

```
mamba install jax
Could not solve for environment specs
The following packages are incompatible
└─ jax   is uninstallable because there are no viable options
   ├─ jax [0.1.39|0.1.40|...|0.3.7] would require
   │  └─ jaxlib  , which does not exist (perhaps a missing channel);
   ├─ jax 0.3.14 would require
   │  └─ jaxlib >=0.3.7 , which does not exist (perhaps a missing channel);
   ├─ jax [0.3.15|0.3.16|...|0.3.20] would require
   │  └─ jaxlib >=0.3.14 , which does not exist (perhaps a missing channel);
   ├─ jax [0.3.21|0.3.22|0.3.23] would require
   │  └─ jaxlib >=0.3.15 , which does not exist (perhaps a missing channel);
   ├─ jax [0.3.24|0.3.25|0.4.1] would require
   │  └─ jaxlib >=0.3.22 , which does not exist (perhaps a missing channel);
   ├─ jax [0.4.10|0.4.11|0.4.8|0.4.9] would require
   │  └─ jaxlib >=0.4.7 , which does not exist (perhaps a missing channel);
   ├─ jax 0.4.12 would require
   │  └─ jaxlib >=0.4.11 , which does not exist (perhaps a missing channel);
   ├─ jax 0.4.2 would require
   │  └─ jaxlib >=0.4.1 , which does not exist (perhaps a missing channel);
   ├─ jax [0.4.3|0.4.4] would require
   │  └─ jaxlib >=0.4.2 , which does not exist (perhaps a missing channel);
   ├─ jax [0.4.5|0.4.6] would require
   │  └─ jaxlib >=0.4.4 , which does not exist (perhaps a missing channel);
   └─ jax 0.4.7 would require
      └─ jaxlib >=0.4.6 , which does not exist (perhaps a missing channel).

```

and 

```
mamba install jaxlib

Could not solve for environment specs
The following package could not be installed
└─ jaxlib   does not exist (perhaps a typo or a missing channel).
```

ChrisDaunt1984, 2023-06-19:
I did get this to work after a lot of trial and error some weeks ago, what exactly worked are still a bit hazy though.
I did manually download the wheel associated my python version (3.9) and installed it via pip that way. I remember having to then downgrade my nvidia drivers too at the same time to install the cuda variants.

let me know if there is anything else that would help you debug. I know this issue is putting some off using it as it is seen as "linux only"

The resulting environment is as follows:
```
# Name                    Version                   Build  Channel      
absl-py                   1.4.0                    pypi_0    pypi       
alabaster                 0.7.13             pyhd8ed1ab_0    conda-forge
ansiwrap                  0.8.4                      py_0    conda-forge
antlr-python-runtime      4.9.3              pyhd8ed1ab_1    conda-forge
anyio                     3.6.2              pyhd8ed1ab_0    conda-forge
appdirs                   1.4.4              pyh9f0ad1d_0    conda-forge
argon2-cffi               21.3.0             pyhd8ed1ab_0    conda-forge
argon2-cffi-bindings      21.2.0           py39ha55989b_3    conda-forge
asttokens                 2.2.1              pyhd8ed1ab_0    conda-forge
attrs                     21.4.0             pyhd8ed1ab_0    conda-forge
babel                     2.11.0             pyhd8ed1ab_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                pyhd8ed1ab_3    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
beautifulsoup4            4.11.1             pyha770c72_0    conda-forge
black                     22.12.0          py39hcbf5309_0    conda-forge
bleach                    6.0.0              pyhd8ed1ab_0    conda-forge
blosc                     1.21.3               hdccc3a2_0    conda-forge
blosc2                    2.0.0                    pypi_0    pypi
bokeh                     3.0.3              pyhd8ed1ab_0    conda-forge
brotli                    1.0.9                hcfcfb64_8    conda-forge
brotli-bin                1.0.9                hcfcfb64_8    conda-forge
brotlipy                  0.7.0           py39ha55989b_1005    conda-forge
build                     0.7.0              pyhd8ed1ab_0    conda-forge
bump2version              1.0.1              pyh9f0ad1d_0    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2023.01.10           haa95532_0
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cairo                     1.16.0            hd694305_1014    conda-forge
certifi                   2022.12.7          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1           py39h68f70e3_3    conda-forge
cfgv                      3.3.1              pyhd8ed1ab_0    conda-forge
charset-normalizer        2.1.1              pyhd8ed1ab_0    conda-forge
chex                      0.1.5                    pypi_0    pypi
click                     8.1.3           win_pyhd8ed1ab_2    conda-forge
click-completion          0.5.2            py39hcbf5309_5    conda-forge
click-log                 0.4.0              pyhd8ed1ab_0    conda-forge
cmarkgfm                  0.8.0            py39ha55989b_2    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
comm                      0.1.2              pyhd8ed1ab_0    conda-forge
commonmark                0.9.1                      py_0    conda-forge
contourpy                 1.0.7            py39h1f6ef14_0    conda-forge
coverage                  7.1.0            py39ha55989b_0    conda-forge
cryptography              39.0.0           py39hb6bd5e6_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
cython                    0.29.33                  pypi_0    pypi
dataclasses               0.8                pyhc8e2a94_3    conda-forge
debugpy                   1.6.6            py39h99910a6_0    conda-forge
decorator                 5.1.1              pyhd8ed1ab_0    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distlib                   0.3.6              pyhd8ed1ab_0    conda-forge
dm-tree                   0.1.8                    pypi_0    pypi
docutils                  0.17.1           py39hcbf5309_3    conda-forge
entrypoints               0.4                pyhd8ed1ab_0    conda-forge
etils                     1.0.0                    pypi_0    pypi
exceptiongroup            1.1.0              pyhd8ed1ab_0    conda-forge
execnet                   1.9.0              pyhd8ed1ab_0    conda-forge
executing                 1.2.0              pyhd8ed1ab_0    conda-forge
expat                     2.5.0                h1537add_0    conda-forge
fastcore                  1.5.27                   pypi_0    pypi
fastrelease               0.1.17                     py_0    fastai
[... 301 lines cut ...]
widgetsnbextension        3.6.1              pyha770c72_0    conda-forge
win32_setctime            1.1.0              pyhd8ed1ab_0    conda-forge
win_inet_pton             1.1.0              pyhd8ed1ab_6    conda-forge
wincertstore              0.2             pyhd8ed1ab_1009    conda-forge
winpty                    0.4.3                         4    conda-forge
[... cut ...]
xyzservices               2022.9.0           pyhd8ed1ab_0    conda-forge
xz                        5.2.6                h8d14728_0    conda-forge
yaml                      0.2.5                h8ffe710_2    conda-forge
zeromq                    4.3.4                h0e60522_1    conda-forge
zipp                      3.12.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               hcfcfb64_4    conda-forge
zstd                      1.5.2                h12be248_6    conda-forge
```

tvt173, 2023-06-23:
looks like jax and jaxlib somewhat-officially support windows now:
https://github.com/google/jax/issues/5795#issuecomment-1603482951

hopefully this helps to make the process easier!

joamatab, 2023-06-23:
amazing!

https://github.com/gdsfactory/gdsfactory/pull/1804

this will simplify the sax installation on windows

@dominauta

joamatab, 2023-08-28:
JAX on windows is working now :)

closed 2023-08-28 by joamatab