gdsfactory/sax #29: jax update breaks sax installation

cpascual, 2024-02-12:
With jax update to 0.4.24, the installation of sax is broken:

```
$ pip install sax
   ...
$ python -c 'import sax' 
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/carlos/mambaforge/envs/swtchd/lib/python3.11/site-packages/sax/__init__.py", line 19, in <module>
    from . import backends as backends
  File "/home/carlos/mambaforge/envs/swtchd/lib/python3.11/site-packages/sax/backends/__init__.py", line 40, in <module>
    from .klu import analyze_circuit_klu, analyze_instances_klu, evaluate_circuit_klu
  File "/home/carlos/mambaforge/envs/swtchd/lib/python3.11/site-packages/sax/backends/klu.py", line 9, in <module>
    import klujax
  File "/home/carlos/mambaforge/envs/swtchd/lib/python3.11/site-packages/klujax.py", line 164, in <module>
    @xla_register_cpu(coo_mul_vec_c128, klujax_cpp.coo_mul_vec_c128)
     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/carlos/mambaforge/envs/swtchd/lib/python3.11/site-packages/klujax.py", line 64, in decorator
    xla.backend_specific_translations["cpu"][primitive] = partial(fun, name)
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'jax.interpreters.xla' has no attribute 'backend_specific_translations'
```

Not sure if this should be fixed at sax level (by pinning jax and jaxlib to 0.4.23 as [here](https://github.com/gdsfactory/gplugins/pull/332)) of it it should be fixed in `klujax`

I can confirm that using `jax==0.4.23` and `jaxlib==0.4.23` works fine.

flaport, 2024-02-13:
I'll probably have to fix klujax. I'll get back when it's done.

flaport, 2024-02-20:
Fixed in 0.11.4.

closed 2024-02-20 by flaport