gdsfactory/sax #31: `IndexError`

tylerflex, 2024-03-07:
Hi Floris, we have a [notebook](https://docs.flexcompute.com/projects/tidy3d/en/latest/notebooks/AdjointPlugin11CircuitMZI.html) demonstrating sax. It seems to error at cell [18] for some users and not for others. There seem to be only minor differences in the dependencies and we can't figure out what is causing this discrepancy. 

Do you have any suggestions for things to look into here? We're stumped after testing several different dependencies.

This is the stack trace

```
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
Cell In[17], line 2
      1 # how to pass specific parmaeters to each of the sub-functions for the instances
----> 2 s = circuit_fn(splitter={"params": params0}, combiner={"params": 0 * params0}, beta=3, phase_sifter=dict(phi=2.0))

File ~/miniconda3/envs/flex/lib/python3.11/site-packages/sax/saxtypes.py:267, in sdict.<locals>.wrapper(**kwargs)
    265 @functools.wraps(model)
    266 def wrapper(**kwargs):
--> 267     return sdict(model(**kwargs))

File ~/miniconda3/envs/flex/lib/python3.11/site-packages/sax/circuit.py:226, in _flat_circuit.<locals>._circuit(**settings)
    223 for inst_name, model in inst2model.items():
    224     instances[inst_name] = model(**full_settings.get(inst_name, {}))
--> 226 S = evaluate_fn(analyzed, instances)
    227 return S

File ~/miniconda3/envs/flex/lib/python3.11/site-packages/sax/backends/klu.py:122, in evaluate_circuit_klu(analyzed, instances)
    117     idx += len(ports_map)
    119 Sx = jnp.concatenate(
    120     [jnp.broadcast_to(sx, (*batch_shape, sx.shape[-1])) for sx in Sx], -1
    121 )
--> 122 CSx = Sx[..., mask]
    123 Ix = jnp.ones((*batch_shape, n_col))
    124 I_CSx = jnp.concatenate([-CSx, Ix], -1)

File ~/miniconda3/envs/flex/lib/python3.11/site-packages/jax/_src/array.py:319, in ArrayImpl.__getitem__(self, idx)
    317   return lax_numpy._rewriting_take(self, idx)
    318 else:
--> 319   return lax_numpy._rewriting_take(self, idx)

File ~/miniconda3/envs/flex/lib/python3.11/site-packages/jax/_src/numpy/lax_numpy.py:4152, in _rewriting_take(arr, idx, indices_are_sorted, unique_indices, mode, fill_value)
   4146     if (isinstance(aval, core.DShapedArray) and aval.shape == () and
   4147         dtypes.issubdtype(aval.dtype, np.integer) and
   4148         not dtypes.issubdtype(aval.dtype, dtypes.bool_) and
   4149         isinstance(arr.shape[0], int)):
   4150       return lax.dynamic_index_in_dim(arr, idx, keepdims=False)
-> 4152 treedef, static_idx, dynamic_idx = _split_index_for_jit(idx, arr.shape)
   4153 return _gather(arr, treedef, static_idx, dynamic_idx, indices_are_sorted,
   4154                unique_indices, mode, fill_value)

File ~/miniconda3/envs/flex/lib/python3.11/site-packages/jax/_src/numpy/lax_numpy.py:4230, in _split_index_for_jit(idx, shape)
   4226   raise TypeError(f"JAX does not support string indexing; got {idx=}")
   4228 # Expand any (concrete) boolean indices. We can then use advanced integer
   4229 # indexing logic to handle them.
-> 4230 idx = _expand_bool_indices(idx, shape)
   4232 leaves, treedef = tree_flatten(idx)
   4233 dynamic = [None] * len(leaves)

File ~/miniconda3/envs/flex/lib/python3.11/site-packages/jax/_src/numpy/lax_numpy.py:4552, in _expand_bool_indices(idx, shape)
   4550     expected_shape = shape[start: start + _ndim(i)]
   4551     if i_shape != expected_shape:
-> 4552       raise IndexError("boolean index did not match shape of indexed array in index "
   4553                        f"{dim_number}: got {i_shape}, expected {expected_shape}")
   4554     out.extend(np.where(i))
   4555 else:

IndexError: boolean index did not match shape of indexed array in index 1: got (18,), expected (14,)
```

And when we `pip show` for the erroring case (python 3.11 on ubuntu)

```
[... 232 lines cut ...]
toolz                         0.12.1
tornado                       6.4
tox                           4.12.1
tqdm                          4.66.1
traitlets                     5.14.1
trimesh                       3.20.0
typeguard                     2.13.3
types-python-dateutil         2.8.19.20240106
typing_extensions             4.9.0
typing-inspect                0.9.0
typing-utils                  0.1.0
tzdata                        2023.4
uri-template                  1.3.0
urllib3                       1.26.18
virtualenv                    20.25.0
vtk                           9.2.6
wcwidth                       0.2.13
webcolors                     1.13
webencodings                  0.5.1
websocket-client              1.7.0
wheel                         0.41.2
widgetsnbextension            4.0.9
xarray                        2023.12.0
zipp                          3.17.0
```

flaport, 2024-03-09:
I was able to reproduce the issue. The good news is that I've seen the error before. The bad news is that I haven't been able to fully understand it yet. I'll keep looking.

flaport, 2024-03-09:
Ok, I figured out the issue.

One of the optimizations SAX makes is to assume that the shape of (i.e. the ports of) an S-matrix generated by a model function never changes. However, your component function takes a shape argument and hence the output shape of the s-matrix can vary. Unfortunately a partial is not going to save you from that because of some more introspection logic that SAX does to construct the circuit.

An easy workaround is to rewrite your partials as actual functions. Something like this:

```python
def component1x2(params=params0, beta=5):
    return component(params=params, beta=beta, shape=(1,2))
    
def component2x2(params=params0, beta=5):
    return component(params=params, beta=beta, shape=(2,2))

circuit_fn, _ = sax.circuit(
    netlist={
        "instances": {
            "splitter": component1x2,
            "phase_shifter": phase_shifter,
            "combiner": component2x2,
        },
        "connections": {
            "splitter,out0": "phase_shifter,in",
            "phase_shifter,out": "combiner,in0",
            "splitter,out1": "combiner,in1",
        },
        "ports": {
            "in": "splitter,in0",
            "out0": "combiner,out0",
            "out1": "combiner,out1",
        },
    }
)

circuit_fn
```

This solves the issue.

I will work on better error messages to handle this case in a future release.

closed 2024-03-09 by flaport

tylerflex, 2024-03-09:
Thanks! Really appreciate you looking into that. Out of curiosity why do you think it works on some environments and not others? Even if sax and jax are the same requirements?

flaport, 2024-03-09:
Not really sure... I seem to have the problem in all my sax environments... Maybe at some point you were optimizing a 2x2x2x2 in stead of a 1x2x2x2?

flaport, 2024-03-10:
One thing I just realized is that it might also be related to whether you have `klujax` installed or not. In the case when klujax is installed SAX will default to the significantly faster (at least for large circuits) KLU backend (`backend='klu'` in `sax.circuit`) rather than the alternative approach by [Gunnar Filipsson](https://ieeexplore.ieee.org/document/4131699) (`backend='fg'` in `sax.circuit`). The latter backend has less strict requirements on the shapes of the models but the implementation in SAX is generally speaking slower (although I doubt that's the case for the circuit in your notebook as that one is very small)

tylerflex, 2024-03-10:
Interesting. I'm pretty sure klujax was installed in both environments with the same version number but not 100% positive