gdsfactory/sax #47: Memory Problem

softbai, 2025-03-31:
Hi Flatport,
I am sorry to bother you again. After getting the S parameter, the old memory is not freed. Deleting variables doesn't help either. This is very detrimental to multiple simulations in some situation. You can try the code as following:

`
#%%
import jax
import jax.numpy as jnp
import matplotlib.pyplot as plt
import sax
#%%
loss = 0.1  # [dB/μm] (alpha) waveguide loss
neff = 2.34  # Effective index of the waveguides
ng = 3.4  # Group index of the waveguides
wl0 = 1.55  # [μm] the wavelength at which neff and ng are defined
ring_length = 10.0  # [μm] Length of the ring
coupling = 0.5  # [] coupling of the coupler
wl = jnp.linspace(1.5, 1.6, 10001)  # [μm] Wavelengths to sweep over
#%%
all_pass_sax, _ = sax.circuit(
    netlist={
        "instances": {
            "dc": {"component": "coupler", "settings": {"coupling": coupling}},
            "top": {
                "component": "straight",
                "settings": {
                    "length": ring_length,
                    "loss": loss,
                    "neff": neff,
                    "ng": ng,
                    "wl0": wl0,
                    "wl": wl,
                },
            },
        },
        "connections": {
            "dc,out1": "top,in0",
            "top,out0": "dc,in1",
        },
        "ports": {
            "in0": "dc,in0",
            "out0": "dc,out0",
        },
    },
    models={
        "coupler": sax.models.coupler,
        "straight": sax.models.straight,
    },
)
#%%
all_pass_sax_jitted = jax.jit(all_pass_sax)
for i in range(1000):
    all_pass_sax_jitted()
#%%

`

softbai, 2025-04-03:
It seems to be happening in klujax c++ code

softbai, 2025-04-12:
Sorry, I am new in github. Thank you very much!

flaport, 2025-05-22:
This was fixed in klujax>=0.4.2

closed 2025-05-22 by flaport